adafruit/Adafruit_CircuitPython_Motor #2: Incorporate DC Motor Characteristics as Optional Parameters

CedarGroveStudios, 2018-03-19:
1) Abstracting duty cycle to a -1.0 to +1.0 motor speed parameter could be improved to match actual motor speed by including the spin threshold voltage of the target DC motor. Since a DC motor's spin threshold voltage can be 10% to 20% of the DC motor's voltage range (1 to 2 volts for a 6-volt motor), a noticeable dead zone exists as the speed parameter value approaches zero. For example, the motor may stop spinning when the PWM duty cycle is less than 20%. The spin threshold is motor dependent and can be derived from the motor datasheet or through experimentation. Incorporating spin threshold would require that the motor controller's supply voltage be known by the driver.
2) Limiting maximum duty cycle to match the DC motor's maximum voltage would allow the use of motors with maximum ratings lower than the motor controller's power source, while maintaining a full -1.0 to +1.0 motor speed parameter range. When powering a 3-volt motor from a 5-volt power source, for example, the speed parameter value for maximum speed would be +1.0 for the forward direction. This would also prevent overdriving the motor and/or motor controller board. The maximum voltage is motor dependent and can be derived from the motor datasheet or through experimentation. Incorporating a maximum voltage limit would require that the motor controller's supply voltage be known by the driver.

tannewt, 2018-03-20:
Both good ideas! This sounds very similar to the tuning that can be done for servo timing. Perhaps we could add `min_duty_cycle` and `max_duty_cycle` kwargs to the constructor. This would be similar to `min_pulse` and `max_pulse` for servo.

CedarGroveStudios, 2018-03-20:
That would certainly help provide trim values for a larger variety of DC motors and would be an economic solution. Consider the next level of abstraction where duty_cycle is just an internal function for speed control rather than directly related to the driver's motor speed parameter value. 
In the experimental scheme I'm testing now, the motor speed parameter value is directly related to the specific motor and power source. For example, a 6-volt motor with a 2 volt spin threshold operating from a 9-volt battery power source has a ```stop_duty_cycle``` of 22% and ```max_duty_cycle``` of 67%. When the battery voltage decays to 8 volts, ```stop_duty_cycle``` and ```max_duty_cycle``` values change to 25% and 75% to continue to observe the spin threshold and optimum (maximum) voltages of the motor (2 volts and 6 volts). Ultimately, a motor speed parameter value of 0 would be just below the spin threshold of the motor; a value of +1.0 would be the full rated speed of the motor, regardless of the motor controller's power supply voltage.

tannewt, 2018-03-21:
Ah! That could be done inside the class or on the outside by a battery monitoring routine.

CedarGroveStudios, 2018-03-21:
Agree. Including duty_cycle trim value kwargs would be a useful feature.

I'll eventually have some examples of the battery and motor attribute -centric approach with an accel/decel (simulated inertia) module. Probably will post those as part of a model train controller learning guide.

CedarGroveStudios, 2018-04-02:
Modified the simulated inertia algorithm from the DC motor version used in the string car to work with x-y servos. Sent a copy to John Park to try as a movement smoothing approach for his recent workshop pan/tilt project.
Code snippet is here [galvanometer](https://github.com/CedarGroveStudios/Galvanometer/blob/master/2018-03-30%20servo%20inertia%20test%20v00.py)

closed 2018-04-20 by CedarGroveStudios