adafruit/Adafruit_CircuitPython_Motor #4: Add Micro Servo (#169) to servo_sweep.py example

CedarGroveStudios, 2018-04-20:
It would be very useful for the min_pulse and max_pulse parameters for the TowerPro SG92R Micro Servo (PID# 169) to be included in the servo_sweep.py example. The following values were determined through experimentation with two servos purchased more than 12 months apart. Here's a suggested addition for the example:

```
# This is an example for the Micro servo - TowerPro SG-92R:
#   https://www.adafruit.com/product/169
# servo7 = servo.Servo(pca.channels[7], min_pulse=500, max_pulse=2400)
```

The current product page lists the values incorrectly as min_pulse=1000usec and max_pulse=2000us. I will submit a correction via the forums.

tannewt, 2018-04-20:
Mind creating a PR by clicking the pencil icon next to history here: https://github.com/adafruit/Adafruit_CircuitPython_Motor/blob/master/examples/servo_sweep.py ?

CedarGroveStudios, 2018-04-20:
Have never initiated a PR yet. Here we go.

tannewt, 2018-04-20:
Thanks! Fixed by https://github.com/adafruit/Adafruit_CircuitPython_Motor/commit/6da5d14b98ac3d929662731f73ce03c0c958b520

closed 2018-04-20 by tannewt