adafruit/Adafruit_CircuitPython_Motor #6: Naming and defaults for steppers could be changed

deanm1278, 2018-06-25:
Hi!
I ran into some confusing when trying to work with the stepper library.
I found the naming of the different drive methods to be confusing. I had assumed SINGLE, the default option would be full-step drive (the motor moves a single step), but instead it's what most other references calls "wave drive" (see https://en.wikipedia.org/wiki/Stepper_motor, http://www.ti.com/lit/an/slva767a/slva767a.pdf). What most other references call full-step, the library calls DOUBLE. Additionally, that cited app note also mentions that:

> The Wave Drive mode of operation is typically considered to be the most simple way of driving a stepper motor, but is also the least common mode of stepper motor driving, due to its shortfalls relative to Full-Step (torque) and Half-step (precision) patterns

I did notice much higher vibration and higher precision waveforms were needed when trying to drive motors this way.

My proposed solution to improve the library and cut down on confusion would be to rename DOUBLE to FULL_STEP and make this the default, then rename SINGLE to WAVE and INTERLEAVE to HALF_STEP.

closed 2018-06-25 by deanm1278

dhalbert, 2018-06-25:
@deanm1278 Did you mean to close this?

ladyada, 2018-06-25:
yeah we discussed and agreed that my naming is Wrong but its been Wrong for 8 years and he's ok to live with it :)