adafruit/Adafruit_CircuitPython_Motor #11: Minor: print statements in Example code fail on motor4.throttle

larryfast, 2018-07-01:
In my test environment "motor4.throttle" fails.
examples/dc_motor.py: line 35
      print("throttle:", motor4.throttle)
Looks like the getter (@properties) code isn't working as expected

Thanks for maintaining excellent demo code!

My environment:
Adafruit HUZZAH
CircuitPython version: adafruit-circuitpython-feather_huzzah-2.3.1.bin
Bundle: adafruit-circuitpython-bundle-2.3.1-mpy-20180630.zip

Bundle downloaded by ampy ... put lib

tannewt, 2018-07-02:
How is it failing?

dhalbert, 2018-07-02:
I traced this down:
```
>>> motor4.throttle
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_motor/motor.py", line 60, in throttle
  File "adafruit_pca9685.py", line 75, in duty_cycle
  File "adafruit_register/i2c_struct_array.py", line 71, in __getitem__
TypeError: function does not take keyword arguments
```
The problem is actually that `struct.unpack_from(format, buffer, offset=0)` as implemented in `shared-bindings/struct/__init__.c` does not allow `offset` as a keyword argument, even though `adafruit_register/i2c_struct_array.py` calls it that way:
```
        return struct.unpack_from(self.format, buf, offset=1)
```
So this is a CircuitPython bug.

dhalbert, 2018-07-02:
This needs to be fixed by fixing https://github.com/adafruit/circuitpython/issues/984. Thanks for finding this!

closed 2018-08-02 by tannewt