adafruit/Adafruit_CircuitPython_Motor #17: provide a way to turn off a Servo

dhalbert, 2018-11-08:
`Servo` doesn't have a way of setting the duty cycle to zero, so that the servo becomes unpowered completely. In Arduino land, `detach()` does that, because it removes the connection to the (hidden) pwm object (if I understand correctly).

@caternuson was trying to translate the first example here: https://learn.adafruit.com/analog-feedback-servos/servos-as-input-devices and encountered the API differences.

Proposal: add a `turn_off()` method, which sets the `PWMOut` duty cycle to zero. Setting the `.fraction` to any value after that will repower the servo to some non-zero duty cycle.

Tagging @tannewt, @TheKitty, and @ladyada here for any comments they may have.

ladyada, 2018-11-08:
yep agree. this is because 'zero' in servo land is 0.5ms pulse, not actually off

caternuson, 2018-11-08:
Or via an `enabled` property?
```python
>>> servo.enabled
True
>>> servo.enabled = False  # turns it "off"
>>> servo.enabled = True   # turns it back "on"
```

dhalbert, 2018-11-08:
If there's an `.enabled` property, would setting `.fraction` re-enable it? That could be fine, but it might be unexpected. Or if could ignore setting `.fraction` if it's disabled, but I'm not sure I like that, because `.enabled` seems like a general thing that could apply to more than `Servo`, but I'm not sure it makes sense to add to everything in this library.

ladyada, 2018-11-08:
i think no, you should be able to have enabled on/off while changing .fraction without it moving. once its disabled, it should not move!

JohnOmernik, 2019-05-16:
IS there any update on this. This is a big difference between the Python2 Adafruit libraries and the Circuit Python Python3 libraries. We need a parity in features, or at least a work around.

JohnOmernik, 2019-05-16:
What about allowing -1 to actuation range as the angle?  0 is no angle, -1 is off with no change?
If anything other than -1 to actuation range then throw an error. 

From the angle_setter

if new_angle < -1 or new_angle > self.actuation_range:
    raise ValueError("Angle out of range")
elif new_angle == -1:
    # Insert code to set pwm.(channel, 0, 0)
else
    self.fraction = new_angle / self.actuation_range

This would allow us to simply issue an angle command, and then issue a -1 without changing the last angle set. It also doesn't "disable" or anything the servo, it's just ready for the next command.

ladyada, 2019-05-16:
added in https://github.com/adafruit/Adafruit_CircuitPython_Motor/pull/27

caternuson, 2019-05-16:
fixed by #27

closed 2019-05-16 by caternuson