adafruit/Adafruit_CircuitPython_Motor #18: Remove methods on DCMotor and ContinuousServo

dhalbert, 2018-11-08:
`DCMotor` and `ContinuousServo`, but not regular `Servo`, provide a `deinit()` operation that sets `.throttle` to 0. However, they don't actually deinit anything: you could change the speed again in the future.

I'm not sure why they were added (they were in from the start). I think we could just delete them.

Came up while @caternuson was recoding an Arduino example (see #17), but this is a separate issue.

Tagging @tannewt for any comment.

tannewt, 2018-11-12:
I'm ok if you want to remove it. They are there to match how you use other CircuitPython classes. Either way is fine.

geekguy-wy, 2020-02-14:
I will remove them. PR #37 fixes this

FoamyGuy, 2020-02-25:
This PR was merged, closing the issue.

closed 2020-02-25 by FoamyGuy