adafruit/Adafruit_CircuitPython_Motor #41: Add method to manually set pulse width

kelliott121, 2020-04-30:
On the Arduino, the writeMicroseconds method of the Servo class is useful for calibrating servos that don't align directly with the 1-2ms range. An equivalent function or field in the servo class would be more convenient than setting the pulse width range to a wide value and calculating the actual limits.

caternuson, 2020-05-01:
Could you accomplish the same thing by tweaking the `min_pulse` and `max_pulse` values via `set_pulse_width_range`?

kelliott121, 2020-05-02:
> Could you accomplish the same thing by tweaking the `min_pulse` and `max_pulse` values via `set_pulse_width_range`?

You could, but I figured it would be easier to set the range really wide and then manually set the duty_cycle usec to find the edges.

caternuson, 2020-05-04:
So this is most useful for an initial "calibration" of a given servo? You'd open up the range, manually set duty cycle to find the min/max and also determine the angle range, then you would use those results to set range and min/max. After that you'd use as normal, i.e. via the `angle` property?

tekktrik, 2022-03-17:
Looks like this won't fit on the Circuit Playground Express + Crickit build w/ Russian translation, so unfortunately this won't work.  A separate library may be a better option, so I'll close this for now.

closed 2022-03-17 by tekktrik