adafruit/Adafruit_CircuitPython_Motor #53: Recommend adding DCMotor controller mode getter/setter

CedarGroveStudios, 2021-03-04:
Thanks to a wonderful question/assertion from @ZodiusInfuser (thank you), I experimented with a change of DCMotor's default "coasting" (fast) decay mode to "braking" (slow) decay mode. The result was even more significant than adjusting PWM frequencies for expanding the controllable motor speed range. Braking mode also linearized the relationship between throttle and motor speed as well as reducing sensitivity to higher (>500Hz) PWM frequencies. When coupled with a lower PWM frequency such as 25Hz, braking mode increased torque at low motor speeds and contributed to a very low spin threshold.

The braking mode change was successfully tested with TB6612 and DRV8833-based motor controllers (Crickit, MotorWing, and various breakouts), but hasn't been tested on the entire range of Adafruit motor controller boards and discrete h-bridge ICs, but issues aren't anticipated. I placed an order to obtain the rest of the Adafruit motor controller inventory and plan to complete the tests when the devices arrive late next week.

I anticipate submitting a PR to add a getter/setter to DCMotor to allow switching between braking and coasting modes on a motor-by-motor basis, keeping coasting as the default for compatibility with existing projects. Would like to have a discussion about the UI for the controller mode. During testing I've been using:
```python
motor1.decay_mode = 0  # Coasting mode
motor1.decay_mode = 1  # Braking mode
```

[image]
Result of mode comparison test at 25Hz. Note the improved spin threshold (speed = red line)  and overall linearity of speed versus throttle (shown as PWM Equivalent Voltage).

CedarGroveStudios, 2021-03-04:
[image]
Result of mode comparison test at 1600Hz (the MotorWing/Shield's default PWM frequency). Note the improved spin threshold (1.6v versus 3.3v).

CedarGroveStudios, 2021-03-08:
Another alternative to ponder would be to eliminate the need for a mode selection parameter by changing the default mode from coasting to braking as recommended by TI in the DRV8871 data sheet. Discrete h-bridges wouldn't be impacted by the change. Modern controller chips like DRV8833, DRV8871, and TB6612 would see dramatically improved performance, but legacy projects with code to linearize the throttle/speed curve may experience a change in speed behavior and battery consumption. 

[image]

ladyada, 2021-03-08:
sure, wanna send over a PR?

ZodiusInfuser, 2021-03-08:
@CedarGroveStudios My view is both modes should be available, as there could be the odd cases where for example you've got a motor spinning and want it to coast to a stop. That does however require a change to the current behavior though, which as we discussed privately does coasting for driving, and braking for stopping.

CedarGroveStudios, 2021-03-08:
@ZodiusInfuser, @ladyada:  I propose that we 1) add the decay_mode getter/setter and 2) change the throttle value interpretation for braking and coasting to be more consistent with modern motor controllers. Although it may impact existing implementations, I believe it makes more sense for a throttle value of zero to cause braking to that speed; None is more logically the high-impedance coasting mode. In either case, discrete h-bridge implementations should see no operational change. Thoughts?

[image]

CedarGroveStudios, 2021-03-08:
@ladyada I'm planning to submit a PR after testing with all the Adafruit boards and the pertinent portion of the motor collection. Have to pace this with my maker budget limitations, so it may take a week or two to complete. I'll rewrite the learning guide after the PR is merged.

closed 2021-03-14 by CedarGroveStudios