adafruit/Adafruit_CircuitPython_Motor #58: ServoKit problem on Jetson Nano

belovictor, 2021-12-27:
Hi,

While using the PCA9685 board with NVIDIA Jetson Nano through I2C I get a mysterious error while trying to run a simple example:
```
import busio
import board
from adafruit_servokit import ServoKit

i2c_bus0=(busio.I2C(board.SCL_1, board.SDA_1))
kit = ServoKit(channels=16, i2c=i2c_bus0)

kit.servo[0].angle = 180
```
Running this code leads to the following error:
```
Traceback (most recent call last):
  File "servo_test.py", line 10, in <module>
    kit.servo[0].angle = 180
  File "/home/belovictor/.local/lib/python3.6/site-packages/adafruit_servokit.py", line 140, in __getitem__
    import adafruit_motor.servo  # pylint: disable=import-outside-toplevel
  File "/home/belovictor/.local/lib/python3.6/site-packages/adafruit_motor/servo.py", line 18, in <module>
    from pwmio import PWMOut
  File "/home/belovictor/.local/lib/python3.6/site-packages/pwmio.py", line 45, in <module>
    raise NotImplementedError("pwmio not supported for this board.")
NotImplementedError: pwmio not supported for this board.
Exiting... 
Cleaning up pins
```

This looks weird as it should not depend on the availability of PWM pins on the Jetson board?
I have the following versions of Adafruit libraries installed on Jetson, all installed through pip:
adafruit-circuitpython-servokit (1.3.6)
adafruit-circuitpython-pca9685 (3.3.9)
adafruit-circuitpython-motor (3.3.2)
Adafruit-Blinka (6.15.0)

Thanks in advance

ladyada, 2021-12-27:
@tekktrik @FoamyGuy can ya change the except to check for NotImplemented not just ImportError :)

closed 2021-12-27 by ladyada

tekktrik, 2021-12-27:
Glad it got resolved!  Thanks for the ping @ladyada and the quick fix @FoamyGuy!

tekktrik, 2021-12-27:
@FoamyGuy is it worth catching NotImplemented in this block for typing across the other libraries?

ladyada, 2021-12-27:
i dont thik you need to back-port it...but maybe going forward!

belovictor, 2021-12-27:
Hi,

I'm afraid it is a bit more complicated :). It's still not working with the new version of the motor library:
```
Traceback (most recent call last):
  File "servo_test.py", line 10, in <module>
    kit.servo[0].angle = 180
  File "/home/belovictor/.local/lib/python3.6/site-packages/adafruit_servokit.py", line 140, in __getitem__
    import adafruit_motor.servo  # pylint: disable=import-outside-toplevel
  File "/home/belovictor/.local/lib/python3.6/site-packages/adafruit_motor/servo.py", line 28, in <module>
    class _BaseServo:  # pylint: disable-msg=too-few-public-methods
  File "/home/belovictor/.local/lib/python3.6/site-packages/adafruit_motor/servo.py", line 36, in _BaseServo
    self, pwm_out: PWMOut, *, min_pulse: int = 750, max_pulse: int = 2250
NameError: name 'PWMOut' is not defined
Exiting... 
Cleaning up pins

```

FoamyGuy, 2021-12-27:
@belovictor can you try version 3.3.1 in your environment: https://github.com/adafruit/Adafruit_CircuitPython_Motor/releases/tag/3.3.1

I'm curious whether that one works for you.

The line raising the exception you posted is here: https://github.com/adafruit/Adafruit_CircuitPython_Motor/blob/fefb7ad03b89fe1478e2146858e43b0d09a403c6/adafruit_motor/servo.py#L36

My understanding is that it should be ignoring the type information during runtime, so I'm not sure why that would raise an error for you.

I am wondering though if whatever you are attempting to do does actually require pwmOut? If that is how you are attempting to drive the servo and your main device doesn't have `pwmio` implemented I'm not sure if you'll be able to drive a servo that way. If you can give version 3.3.1 (linked above) a try and let us know how it goes that will help narrow the issue down to specifically the typing information or something different.

FoamyGuy, 2021-12-27:
I found this SO answer which might be the same root cause. https://stackoverflow.com/a/27461145/507810

Perhaps doing the annotation like this could allow it to work:
```
 self, pwm_out: 'PWMOut', *, min_pulse: int = 750, max_pulse: int = 2250 
```

belovictor, 2021-12-27:
@FoamyGuy after installing version 3.3.1 of motor library with
`pip3 install 'adafruit-circuitpython-motor==3.3.1' --force-reinstall`
my code works without errors, so I would assume this as a workaround for my exact situation :) Thanks a lot!

I'm not using motor library directly, as you may note from the code I've posted in the initial issue, motor library is being called from inside of ServoKit which also happens to be Adafruit code. I'm sure that using PCA9685 over I2C does not require pwmOut though :)

FoamyGuy, 2021-12-27:
Thanks for trying that out and confirming. 

I'll look into this today and see if we can use a string for the type annotation. If that doesn't cause any other issues then it may be the best way to go.

tekktrik, 2021-12-27:
My understanding of using strings as annotations is for when something hasn't been defined yet but will be.  So for example, you're typing as a class that will be defined later in the same file.  I think you will still get errors from a static type checker if you type something in strings that isn't or won't be available in the namespace.

tekktrik, 2021-12-27:
So in this context it will still look for PWMOut to be defined somehow, I believe

PR #59 (2021-12-27): catch NotImplementedError
resolves #58 

allows to run in environments without `pwmio` implemented.