adafruit/Adafruit_CircuitPython_Motor #65: Servo motors are not disabling/releasing after setting angle to None

scprotz, 2022-12-01:
I'm using a Raspberry Pi 4 to control the PCA9685.  I have generic MG996R-style servos connected (2 different brands).  When I set the servo.angle = None, the servos just hold their current position without releasing.

I'm using servokit 1.3.13, pca9685 3.4.6, and motor 3.4.5

(I do have a version on a different raspberry pi that is running servokit 1.3.6, pca9685 3.39 and motor 3.3.2 that does work correctly, though it uses a different brand of servo).

I am not quite sure if this is due to a regression or certain brands of servos not recognizing the angle = None.

Any insight would be appreciated.  I may try to upgrade the second raspberry pi and see if those servos get the issue.  That would help narrow down if this is a per-servo-brand bug or a regression.

Recelis, 2022-12-30:
I have the same issue too. My micro servo motors are disabling correctly but it seems like the bigger servos 1501MG [https://www.pololu.com/product/1057](url) aren't being disabled.

ladyada, 2022-12-30:
do you know that the motors will 'release' when not receiving a 50hz signal because thats the only way to release motors - if your servo doesnt do that, it has nothing to do with this firmware

scprotz, 2022-12-30:
How can we programmatically tell it not to send the 50hz signal then. I would have thought setting angle = None should do something like that. Is there another command?

scprotz, 2023-01-01:
So just a note.  It appears I can set the duty_cycle to nothing and it turns off this type of servo.

There are two ways (which are effectively the same based on the source code in github):

either
kit.servo[0]._pwm_out.duty_cycle  = 0

or
kit.servo[0].fraction = None

Both of these turn off duty cycle 'release' the motor.

scprotz, 2023-01-01:
I misspoke.  This does not  release the motor.  It just stops the duty cycle.  I'll play around and see if there is another way to cut power completely to the servo.

arnaudbore, 2023-03-02:
@scprotz do you have any update on this? It feels like you've been working with this for quite some time. If you have some time can you answer #73  ? Thank you in advance

scprotz, 2023-04-03:
Just wanted to add another comment.  I have tried a number of things and still certain brands of motors will not disconnect/release when I set fraction = None, nor Angle = None.  It appears there is a way to turn off the servo pins on the PCA9685.  On this page: https://learn.adafruit.com/16-channel-pwm-servo-driver/library-reference
It states that (using an Arduino), I can setPWM(1, 0, 4096) to disable the servo pins (and I hope turn off the servo).

Is there a way to do the same with CircuitPython_Motor (or one of the lower level APIs)?

paulfertser, 2025-05-16:
I checked how disabling PWM control works through all the levels, starting with the library and ending with the bits in PCA9685 registers. I also checked the resulting signals with oscilloscope. The conclusion is that setting `angle = None` indeed fully turns off the corresponding output, there's no frequency, just constant 0 V signal going to servos in that state. You can also use `i2cdump` in parallel to see the corresponding register having bit 1 set which means "full OFF" according to the datasheet.

This works nicely as expected with "normal" servos but I faced two different issues with two different no-name micro-servos (which were part of "robotic hand" from aliexpress). They are supposedly "robotic servos with anti-stall features" but there're no corresponding datasheets to check. One set of servos insists on keeping the last set position until fully power-cycled. Another set releases as soon as PWM signal is turned off but they do not "recover" (i.e. do not start following new settings) when it reappears.

The conclusion is: the library really stops emitting any and all frequency (other than zero volts DC) and works as expected. It's just some servos that are weird. Ask your servo manufacturer about ways to release. Or use driver board designs which allow to fully power-cycle the servos (@ladyada , can you please tell why the AOD417 Q1 MOSFET on your board can not be contolled externally, I guess it'd be just one more transistor to let a 3.3 V control signal tie the FET's gate to input voltage to reliably close it?).

ladyada, 2025-05-16:
its designed to be a low-dropout polarity protector not a logic switch

paulfertser, 2025-05-16:
> its designed to be a low-dropout polarity protector not a logic switch

Yes, I understand, thank you. Do you think if it might be a good idea if the next revision of the board would also allow a 3.3 V logic signal to control that FET? Apparently there's enough servos in the wild which really need that "cut all the power to recover" feature. Or am I missing something and there's a better way?