adafruit/Adafruit_CircuitPython_Motor #71: Does this library work with raspberry pi?

padmalcom, 2023-02-03:
Hi, this is more a question than an issue but I think this should be documented.
Does this library work with vanilla python on a raspberry pi? If I understand, it does not because
I need to install this Circuit Python which is a completely separate operating system for the rasberry.
Am I right?

caternuson, 2023-02-03:
You do not install the CircuitPython firmware. For CircuitPython *library* use, you'll install an additional library called Blinka:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/overview

With Blinka installed, this library can be used on a Raspberry Pi and installed via pip.

padmalcom, 2023-02-14:
Thank you, this was the information I wanted. Nevertheless, I can not confirm that installing blinka and then the CircuitPyton work.

closed 2023-02-14 by padmalcom

makermelissa, 2023-02-14:
CircuitPython runs without an operating system and I can see it being a little confusing as it can now run on the Raspberry Pi in lieu of any operating system. As @caternuson mentioned, Blinka is what you want if you are running it on Vanilla Python and it takes the place of many of the CircuitPython's built-in modules to allow the CircuitPython libraries themselves to work. Blinka isn't as complete as CircuitPython, so if you do find any modules that would be useful, you can request them at https://github.com/adafruit/Adafruit_Blinka/issues, but generally we have the basics covered so that libraries work.