adafruit/Adafruit_CircuitPython_Motor #73: Get Angle

arnaudbore, 2023-03-02:
Hi,

I would like to know if there is a way to query the current position (angle) of a servo. I can see there is a setter and a property but no getter. Is there another way to get this information ?

Thank you in advance
Arnaud

caternuson, 2023-03-02:
The property is the getter, ex:
```python
print(servo.angle)
```

However, it is *not* the actual true current position of the servo. It is just the currently commanded position. Unless you are using a special feedback servo, like this:
https://www.adafruit.com/product/1449
there is no way to get actual current position.

scprotz, 2023-03-02:
Depending on the servo, a trick I have used in the past was to just 'move' the servo but don't give it any time to actually perform the  move.  At that point the value for the angle has been set and you can read it.

I think you might be able to do something like:

servo.angle = 90
current_angle = servo.angle
servo.angle = current_angle

Been a bit since I've tried something  like that, but it was a nifty workaround.

caternuson, 2023-03-03:
Closing since this does not appear to be a library issue. It's more like an info request or a how-to. If there is an actual issue with the library, please provide more information on the issue and example code that can be used to demonstrate the issue.

closed 2023-03-03 by caternuson