adafruit/Adafruit_CircuitPython_Motor #75: Motor kit / motor issues

kxagan, 2023-07-04:
Rather strange problem: Feather can't seem to process "from adafruit_motorkit import MotorKit". So far I have checked file names, versions, reinstalled motorkit and motor libraries, ensured they are in the right destination, and tried calling them with "import adafruit_motor" and "import adafruit_motorkit", both of which worked properly. Had little to no trouble getting files for the OLED to work and I followed the same steps, as well as closely following the guide: https://learn.adafruit.com/adafruit-stepper-dc-motor-featherwing/circuitpython. Needless to say this is preventing me from running the stepper motor I'm working with.

caternuson, 2023-07-05:
Please post the full text of any error message(s). Or provide more details on what "can't process" means.

kxagan, 2023-07-06:
The only error indication is given by 2 red blinks on the Feather. For reference the attached code is the whole problem. The fix should be simple, just ensure I have all the files named properly and in the right place, but this has yet to work. For clarity there is no error message or issue on the coding end. 

[image]

caternuson, 2023-07-06:
Probably just a library install issue.

There should be a file named `adafruit_motokit.mpy` in your `CIRCUITPY/lib` folder.

There should **NOT** be any folders named `adafruit_motorkit` in your `CIRCUITPY/lib` folder.

kxagan, 2023-07-06:
I cleaned the library and did another fresh install, finally got it working. Thanks for your help.

closed 2023-07-06 by caternuson